ruundii/bthidhub #4: Crash on startup if HID device doesn't have "input"

zhogov, 2020-08-30:
```
sudo systemctl status remapper.service
```
```
● remapper.service - Remapper
   Loaded: loaded (/lib/systemd/system/remapper.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Sun 2020-08-30 03:14:09 EDT; 19s ago
  Process: 1680 ExecStart=/usr/bin/python3 /home/pi/bthidhub/remapper.py (code=exited, status=1/FAILURE)
 Main PID: 1680 (code=exited, status=1/FAILURE)

Aug 30 03:14:08 bthidhub python3[1680]: Traceback (most recent call last):
Aug 30 03:14:08 bthidhub python3[1680]:   File "/home/pi/bthidhub/remapper.py", line 16, in <module>
Aug 30 03:14:08 bthidhub python3[1680]:     hid_devices = HIDDeviceRegistry(loop)
Aug 30 03:14:08 bthidhub python3[1680]:   File "/home/pi/bthidhub/hid_devices.py", line 118, in __init__
Aug 30 03:14:08 bthidhub python3[1680]:     self.__scan_devices()
Aug 30 03:14:08 bthidhub python3[1680]:   File "/home/pi/bthidhub/hid_devices.py", line 166, in __scan_devices
Aug 30 03:14:08 bthidhub python3[1680]:     inputs = os.listdir('/sys/bus/hid/devices/'+device+'/input')
Aug 30 03:14:08 bthidhub python3[1680]: FileNotFoundError: [Errno 2] No such file or directory: '/sys/bus/hid/devices/0003:046D:C52B.0005/input'
Aug 30 03:14:09 bthidhub systemd[1]: remapper.service: Main process exited, code=exited, status=1/FAILURE
Aug 30 03:14:09 bthidhub systemd[1]: remapper.service: Failed with result 'exit-code'.
```

Here are HID devices registered by **Logitech M187** mouse – there seem to be some symlinked linked device folders. Same crash happens with **Logitech K400+** keyboard/touchpad combo, **Microsoft Mouse 4000**.
I guess these descriptors are pretty hard to untangle, but **let's not crash on startup if we cannot detect devices.**
```
/sys/bus/hid/devices/
├── 0003:046D:4019.0003 -> ../../../devices/platform/soc/20980000.usb/usb1/1-1/1-1:1.1/0003:046D:C52F.0002/0003:046D:4019.0003
├── 0003:046D:C52F.0001 -> ../../../devices/platform/soc/20980000.usb/usb1/1-1/1-1:1.0/0003:046D:C52F.0001
└── 0003:046D:C52F.0002 -> ../../../devices/platform/soc/20980000.usb/usb1/1-1/1-1:1.1/0003:046D:C52F.0002
```
```
/sys/bus/hid/devices/0003:046D:C52F.0001
├── country
├── driver -> ../../../../../../../../bus/hid/drivers/logitech-djreceiver
├── hidraw
│   └── hidraw0
│       ├── dev
│       ├── device -> ../../../0003:046D:C52F.0001
│       ├── power
│       │   ├── autosuspend_delay_ms
│       │   ├── control
│       │   ├── runtime_active_time
│       │   ├── runtime_status
│       │   └── runtime_suspended_time
│       ├── subsystem -> ../../../../../../../../../../class/hidraw
│       └── uevent
├── modalias
├── power
│   ├── autosuspend_delay_ms
│   ├── control
│   ├── runtime_active_time
│   ├── runtime_status
│   └── runtime_suspended_time
├── report_descriptor
├── subsystem -> ../../../../../../../../bus/hid
└── uevent
```
```
/sys/bus/hid/devices/0003:046D:C52F.0002
├── 0003:046D:4019.0003
│   ├── builtin_power_supply
│   ├── country
│   ├── driver -> ../../../../../../../../../bus/hid/drivers/logitech-hidpp-device
│   ├── hidraw
│   │   └── hidraw2
│   │       ├── dev
│   │       ├── device -> ../../../0003:046D:4019.0003
│   │       ├── power
│   │       │   ├── autosuspend_delay_ms
│   │       │   ├── control
[... 215 lines cut ...]
│       │   │   └── runtime_suspended_time
│       │   ├── subsystem -> ../../../../../../../../../../../../class/hwmon
│       │   ├── temp1_label
│       │   ├── temp2_label
│       │   └── uevent
│       ├── manufacturer
│       ├── model_name
│       ├── online
│       ├── power
│       │   ├── autosuspend_delay_ms
│       │   ├── control
│       │   ├── runtime_active_time
│       │   ├── runtime_status
│       │   └── runtime_suspended_time
│       ├── powers -> ../../../0003:046D:4019.0003
│       ├── scope
│       ├── serial_number
│       ├── status
│       ├── subsystem -> ../../../../../../../../../../../class/power_supply
│       ├── type
│       └── uevent
├── report_descriptor
├── subsystem -> ../../../../../../../../../bus/hid
└── uevent
```

ruundii, 2020-08-30:
Thanks for trying it and for such a detailed report. I have made a fix (wrapped in try/except) and pushed it, but won't have a chance to test it properly myself for the next 3 days or so.

zhogov, 2020-08-30:
Thanks for the quick fix, service starts successfully now:
[image]

Fun part is that apparently only one of these HID devices gets picked up that results in touchpad sending keypresses instead of moving mouse and keyboard buttons behaving like they have some modifier pressed all the time (e.g. **R** works like Win+R)

closed 2020-08-30 by zhogov