ruundii/bthidhub #8: Keyboard requires passcode

jrabarton, 2020-10-07:
I’m trying this out with a Microsoft keyboard that requires a passcode, and it doesn’t seem that there is any way I can get it to pair with the raspberry pi

ruundii, 2020-11-11:
is it just that the keyboard has a monitor showing the passcode on its own during the pairing?

mo5h, 2020-11-30:
I was able to get it to connect by sshing into the rasberry pi and using this method :  https://bytee.net/blog/linux-surface-ergonomic-keyboard, however it appears that the keyboard hasn't created a input device under /dev/input and isn't forwarding anything to the host, so presumably there's something else I've got to do to get it to work

tcoulter, 2020-12-21:
Did you ever get this figured out? I'm installing bthidhub on my raspberry pi *right now* and it turns out I have the same keyboard.

tcoulter, 2020-12-21:
I was able to get it paired using the method above, but can't seem to get the keyboard to talk to the host.

[image]

tcoulter, 2020-12-22:
Some more of my experience with the ergonomic keyboard: 

1. As you can see from the image, two keyboards were detected. This comes from two different pairings; it appears that the ergonomic keyboard bumps up the id (`70` -> `71`) every time you pair it with a new device from scratch. (As of this writing, my keyboard currently has the `F0:F3:6F:72:75:B0`, where the id has been bumped up to `72`). I noticed through bluetoothctl that the old addresses were cached and therefore still show up in the list above. Would be great to be able to remove them from the UI as desired. 
2. I was able to get bluetoothctl to ask for the password for the keyboard, as outlined in the link above. Unfortunately, I was only able to get this to happen once. Eventually as I tried to continue to set up the device and try to get the hid working, I paired from scratch multiple times (e.g., started all over) and it no longer prompts for a password, so I can't fully pair the keyboard. I tried removing all of bluetoothctl's cache by hand, and using the `remove` command within bluetoothctl, but it still didn't have any effect. As an aside, I'd love if the hub could expose the password dialog so I don't have to go into bluetoothctl myself.
3. This next one isn't related to the keyboard, but still noteworthy: I couldn't get my Windows 10 Surface paired host to stay as a paired host. If I restarted the hub, it would no longer be recognized by the hub as a paired host. I believe this is because of another prompt from bluetoothctl: When setting up the initial pairing, bluetoothctl outputs an some uuid and asks me if I want to accept the service, and asked me to enter yes or no. The hub won't recognize the host as a paired host without me entering yes in bluetoothctl, so on restart, there's no prompt available for me to enter yes into. It'd be awesome if the hub could expose the prompt like above.

Dreamsorcerer, 2021-05-26:
> I was able to get it paired using the method above, but can't seem to get the keyboard to talk to the host.

At a guess, you probably need to change the sdp_record.xml file to match your keyboard. See my brief description at https://github.com/ruundii/bthidhub/issues/11#issuecomment-841819628

Dreamsorcerer, 2025-01-08:
SDP records are modified automatically now, so things should pretty much "just work". Please try reinstalling the latest version.

closed 2025-01-08 by Dreamsorcerer