ruundii/bthidhub #9: RPi2 connected to a Win10 PC with a USB?

bruno-briner, 2020-11-08:
Hi. Interesting project.

Would it work in a RPi2 connected with a USB-USB cable to a PC running Win10?

If so it would be great tool to remap my Apple Magic Keyboard 2 (A1644) when I am working from home.

ruundii, 2020-11-11:
first, i never tried it on RPI2, but in principle it should work, as afaik RPi2 is more powerful than RPi zero.

the usb-to-usb connection to a machine is not supported at the moment, and I am not sure there is a need. You can connect RPI2 to machine via bluetooth. you may also consider using WinAppleKey

bruno-briner, 2020-11-12:
Thanks for the reply. Unfortunately WinAppleKey is not an option as I don't have admin rights to the Win10 (company laptop).

I tried bthidhub in the RPi2, after all was setup it entered in a loop complaining that BT was missing, which was logical.

I will buy a cheap bluetooth dongle from Aliexpress and test it again.

Another question, assuming that the RP2 works with a BT dongle connecting the RPi2 to Win10 via BT. Can the A1644 be connected to the RPi2 via USB instead of BT?

closed 2021-06-26 by Dreamsorcerer