ruundii/bthidhub #10: Failed to load

blovi8, 2020-11-11:
Hi. I've tried to install this project on multiple Pi Zero W boards using different SD cards and install methods but I always get the same errors when I load the [rpi-ip-address]:8080 web interface...

Failed to load update bluetooth devices list. Unauthorized.
Failed to load connected devices list. Unauthorized

I'm using the default user and password on the pi and can connect devices using bluetoothctl on the pi. I've tried to chmod specific files used for the web interface but that doesn't seem to help.

Any assistance would be greatly appreciated!!

ruundii, 2020-11-11:
hi.

are you using the default install script and install to a pre-existing raspbian image?

blovi8, 2020-11-12:
Hi there. Yes, I've tried it a few different install methods using the default script(s). e.g., editing and running install_windows.bat from a windows machine, running setup.sh in bash on the pi, and also running each step of on_pi_setup.sh on the pi. I always end up getting the same result.

gennymcdonagh, 2020-11-19:
Hi, I'm also having the same issue. Have followed the provided instructions in the /install directory, but when I log into the web interface (with default password 'raspberry') I get the 'Unauthorized' messages mentioned above.

f3rn4nd0d, 2020-11-20:
[image]

Hello, Raspbian installed from scratch and followed the instructions, but this is what I am getting so far. I want to connect a keyboard and a mouse both via USB to the Raspberry and use them as Bluetooth devices in several computers. Thanks!

ruundii, 2020-11-20:
I'll check it now, maybe something has changed with the new raspbian image.
wait for an update

ruundii, 2020-11-20:
after a couple of hours of investigation, I figured that something went wrong with a recent release of aiohttp library and for some reason sessions stopped to work in the app. for now I just point install script to a previous version of aiohttp

iravit, 2020-11-20:
I have the same issue. I noticed there is an issue with the aiohttp_session module, particularly with SimpleCookieStorage
function. The function does not create a session cookie. Which version of the aiohttp library do you recommend ?

toretto460, 2021-01-21:
Hi @ruundii I've the same problem with a different error

[image]

I've installed the latest version (with the `aiohttp==3.6.2`)

Here some logs for investigation
```
pi@bthidhub:~ $ sudo systemctl status remapper.service
● remapper.service - Remapper
   Loaded: loaded (/lib/systemd/system/remapper.service; enabled; vendor preset: enabled)
   Active: active (running) since Thu 2021-01-21 08:37:46 GMT; 42min ago
 Main PID: 511 (python3)
    Tasks: 12 (limit: 881)
   CGroup: /system.slice/remapper.service
           └─511 /usr/bin/python3 /home/pi/bthidhub/remapper.py

Jan 21 09:20:35 bthidhub python3[511]:     resp = await handler(request)
Jan 21 09:20:35 bthidhub python3[511]:   File "/usr/local/lib/python3.7/dist-packages/aiohttp/web_middlewares.py", line 119, in impl
Jan 21 09:20:35 bthidhub python3[511]:     return await handler(request)
Jan 21 09:20:35 bthidhub python3[511]:   File "/usr/local/lib/python3.7/dist-packages/aiohttp_session/__init__.py", line 154, in factory
Jan 21 09:20:35 bthidhub python3[511]:     response = await handler(request)
Jan 21 09:20:35 bthidhub python3[511]:   File "/home/pi/bthidhub/web.py", line 191, in get_bluetooth_devices
Jan 21 09:20:35 bthidhub python3[511]:     return web.Response(text=json.dumps(self.adapter.get_devices()))
Jan 21 09:20:35 bthidhub python3[511]:   File "/home/pi/bthidhub/adapter.py", line 186, in get_devices
Jan 21 09:20:35 bthidhub python3[511]:     if self.adapter is None:
Jan 21 09:20:35 bthidhub python3[511]: AttributeError: 'BluetoothAdapter' object has no attribute 'adapter'
pi@bthidhub:~ $ sudo systemctl status bluetooth.service
● bluetooth.service - Bluetooth service
   Loaded: loaded (/lib/systemd/system/bluetooth.service; enabled; vendor preset: enabled)
   Active: active (running) since Thu 2021-01-21 08:37:41 GMT; 43min ago
     Docs: man:bluetoothd(8)
 Main PID: 368 (bluetoothd)
    Tasks: 1 (limit: 881)
   CGroup: /system.slice/bluetooth.service
           └─368 /usr/libexec/bluetooth/bluetoothd -p time,input,autopair,policy,scanparam,deviceinfoNotifyAccess=main

Jan 21 08:37:41 bthidhub bluetoothd[368]: src/main.c:parse_controller_config() Key file does not have group “Controller”
Jan 21 08:37:41 bthidhub bluetoothd[368]: Starting SDP server
Jan 21 08:37:41 bthidhub bluetoothd[368]: Ignoring (cli) hostname
Jan 21 08:37:41 bthidhub bluetoothd[368]: Ignoring (cli) wiimote
Jan 21 08:37:41 bthidhub bluetoothd[368]: Ignoring (cli) hog
Jan 21 08:37:41 bthidhub bluetoothd[368]: Ignoring (cli) gap
Jan 21 08:37:41 bthidhub bluetoothd[368]: Ignoring (cli) deviceinfo
Jan 21 08:37:41 bthidhub bluetoothd[368]: Ignoring (cli) battery
Jan 21 08:37:41 bthidhub bluetoothd[368]: Bluetooth management interface 1.14 initialized
Jan 21 08:37:42 bthidhub bluetoothd[368]: Failed to set privacy: Rejected (0x0b)
```
Let me know if I can check other logs

Dreamsorcerer, 2021-06-26:
This is fixed.

closed 2021-06-26 by Dreamsorcerer