ruundii/bthidhub #12: Keyboard Compatibility Mode not working

joseecm-github, 2020-12-09:
Hello.

After install on a newly flashed Raspbian OS and enter to the web interface I have been able to pair with my Windows 10 PC but it does not receive any keypress from the Logitech K400 wireless keyboard attached to the raspberry pi 3.
I have try to enable the keyboard compatibility mode but then appears the error "Could no set compatibility mode. Internal server error".

Any idea to solve the problem?

Thanks in advance and regards.

Dreamsorcerer, 2021-06-26:
If you're still having issues with this, please ssh into the Pi and check the output of `systemctl status remapper`.

Dreamsorcerer, 2025-01-08:
SDP records are modified automatically now, so things should pretty much "just work" (i.e. you probably don't need compatibility mode anymore). Please try reinstalling the latest version.

closed 2025-01-08 by Dreamsorcerer