ruundii/bthidhub #13: UI interface showing keyboard / mouse connection

tcoulter, 2020-12-22:
In setting up the hub I've had a number of device issues. I *think* I have my keyboard paired, or at some point in time had it paired correctly, but I still had issues with the host, at least that was my hypothesis. I'd love some interface within the hub UI showing if a keyboard is A) not include paired correctly (which seems to be there, as the devices list says "paired"), but also B) showing the keys that are pressed as validation that the connection is correct. It'd be similarly cool with a mouse, showing mouse input is working correctly.

Dreamsorcerer, 2025-01-08:
Things should work a lot better now, so would be a good time to reinstall and try the latest version. But, the UI idea would still be useful if anyone wanted to try and implement it.

Dreamsorcerer, 2025-01-08:
https://github.com/Dreamsorcerer/bthidhub/issues/9

closed 2025-01-08 by Dreamsorcerer