ruundii/bthidhub #16: Non-Mac Keyboard + Mouse

SeaDaddytheRed, 2021-02-05:
Hello, First thank you for this idea.  It is exactly what I have been looking for to solve my multiple computers/one keyboard mouse setup.

I am having some problems though.

1. I am using a non-Mac keyboard, 87 key US, and I don't know the shortcut to switch connected devices.
2. No matter what mouse I connect to the pi it shows up as a mouse in the drop down menu of your program but it doesn't work at all on the computer.  I have tried wireless logitech, and my Anker gaming mouse, neither work.
Thanks again for this great idea.  I hope I can get it working!

Dreamsorcerer, 2021-05-11:
Same issue here, the filter removes all 8 byte commands:
https://github.com/ruundii/bthidhub/blob/master/mouse_message_filter.py#L10

These include all the movement events of a mouse...

I commented out those lines and the mouse works, however the vertical axis is 10x more sensitive than the horizontal axis. The result is that moving the mouse up/down is only just usable at the lowest sensitivity, while move left/right results in the cursor jumping up/down as it moves along because it picks up every little movement on that axis.

Dreamsorcerer, 2021-05-16:
OK, I managed to update the sdp_record.xml, at which point the mouse works properly. I don't think there is a "standard" mouse config you can provide in the sdp record, so it would be better to look at dynamically creating one.

Dreamsorcerer, 2021-06-26:
Let's track this in #23, which should resolve the issue.
I've described a workaround here: https://github.com/ruundii/bthidhub/issues/11#issuecomment-841819628

closed 2021-06-26 by Dreamsorcerer