ruundii/bthidhub #17: Magic Keyboard and Magic Trackpad

kellermanns, 2021-02-07:
Does anyone have this running with the Magic Keyboard and Magic Trackpad 2? 

I can get everything to pair and show up in the web interface, I can pair the Raspberry with both of my MacBooks (as Paired Host), but no input is passed on and no Bluetooth Keyboard, Trackpad or Mouse (emulated?) is found in the Settings). I also can’t enable compatibility mode, but I don’t think that’s related. 

Any help, experience, config screenshots, etc. is really appreciated 

Dave

kwatkins, 2021-02-18:
Same, connects but doesn't pass through. Something I wish I knew before trying this out. Nonetheless, this project in theory is a good one and wish I had more time to figure out why it's not passing through.

Dreamsorcerer, 2021-05-11:
I don't know about the keyboard, but mouse could be the same issue as #16.

See if it works when commenting out this code: https://github.com/ruundii/bthidhub/blob/master/mouse_message_filter.py#L10,L11

Dreamsorcerer, 2021-05-26:
Try changing the sdp_record.xml file as mentioned in https://github.com/ruundii/bthidhub/issues/11#issuecomment-841819628

Dreamsorcerer, 2021-06-26:
I think this will be fixed if #23 is implemented, so let's track there.

closed 2021-06-26 by Dreamsorcerer