ruundii/bthidhub #19: Setup fails with 'sudo: unable to resolve host bthidhub: Name or service not known'

sarimarton, 2021-02-28:
Setup ends with a `sudo: unable to resolve host bthidhub: Name or service not known ... FATAL ERROR: Remote side unexpectedly closed network connection` message. See the log below.

I tried it on both a Raspberry Pi Zero W and a Raspberry Pi 4B. The same log is produced. The webapp works though, and it recognizes the connected devices, but it can't control or restart the service, and nothing really works.

```
C:\Users\sarim\repos\bthidhub\install>plink -ssh -v -pw raspberry pi@172.20.10.3 -m setup.sh
Looking up host "172.20.10.3" for SSH connection
Connecting to 172.20.10.3 port 22
We claim version: SSH-2.0-PuTTY_Release_0.74
Remote version: SSH-2.0-OpenSSH_7.9p1 Raspbian-10+deb10u2
Using SSH protocol version 2
No GSSAPI security context available
Doing ECDH key exchange with curve Curve25519 and hash SHA-256 (unaccelerated)
Server also has ecdsa-sha2-nistp256/ssh-rsa host keys, but we don't know any of them
Host key fingerprint is:
ssh-ed25519 255 0d:9f:59:81:b5:2a:21:e5:4c:50:bf:a3:6f:82:86:c1
The server's host key is not cached in the registry. You
have no guarantee that the server is the computer you
think it is.
The server's ssh-ed25519 key fingerprint is:
ssh-ed25519 255 0d:9f:59:81:b5:2a:21:e5:4c:50:bf:a3:6f:82:86:c1
If you trust this host, enter "y" to add the key to
PuTTY's cache and carry on connecting.
If you want to carry on connecting just once, without
adding the key to the cache, enter "n".
If you do not trust this host, press Return to abandon the
connection.
Store key in cache? (y/n) y
Initialised AES-256 SDCTR (AES-NI accelerated) outbound encryption
Initialised HMAC-SHA-256 (unaccelerated) outbound MAC algorithm
Initialised AES-256 SDCTR (AES-NI accelerated) inbound encryption
Initialised HMAC-SHA-256 (unaccelerated) inbound MAC algorithm
Using username "pi".
Sent password
Access granted
Access granted. Press Return to begin session.
Opening main session channel
Opened main channel
Started a shell/command
Get:1 http://archive.raspberrypi.org/debian buster InRelease [32.8 kB]
Get:2 http://raspbian.raspberrypi.org/raspbian buster InRelease [15.0 kB]
Get:3 http://archive.raspberrypi.org/debian buster/main armhf Packages [364 kB]
Get:4 http://raspbian.raspberrypi.org/raspbian buster/main armhf Packages [13.0 MB]
Get:5 http://raspbian.raspberrypi.org/raspbian buster/non-free armhf Packages [104 kB]
Fetched 13.5 MB in 31s (430 kB/s)
Reading package lists...
Reading package lists...
Building dependency tree...
Reading state information...
Calculating upgrade...
The following packages will be upgraded:
  base-files bind9-host bluez-firmware ca-certificates device-tree-compiler
  file firmware-atheros firmware-brcm80211 firmware-libertas
  firmware-misc-nonfree firmware-realtek iproute2 libbind9-161
  libdns-export1104 libdns1104 libgnutls30 libisc-export1100 libisc1100
  libisccc161 libisccfg163 libldap-2.4-2 libldap-common liblwres161
  libmagic-mgc libmagic1 libpam-systemd libraspberrypi-bin libraspberrypi-dev
  libraspberrypi-doc libraspberrypi0 libssl1.1 libsystemd0 libudev1 libzstd1
  openssl python-rpi.gpio raspberrypi-bootloader raspberrypi-kernel
  raspberrypi-sys-mods raspi-config rpi-eeprom rpi.gpio-common sudo systemd
  systemd-sysv tzdata udev unzip
48 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Need to get 148 MB of archives.
After this operation, 4,626 kB of additional disk space will be used.
Get:1 http://archive.raspberrypi.org/debian buster/main armhf libssl1.1 armhf 1.1.1d-0+deb10u5+rpt1 [1,276 kB]
Get:2 http://quantum-mirror.hu/mirrors/pub/raspbian/raspbian buster/main armhf base-files armhf 10.3+rpi1+deb10u8 [70.1 kB]
Get:3 http://quantum-mirror.hu/mirrors/pub/raspbian/raspbian buster/main armhf systemd-sysv armhf 241-7~deb10u6+rpi1 [101 kB]
Get:4 http://quantum-mirror.hu/mirrors/pub/raspbian/raspbian buster/main armhf libpam-systemd armhf 241-7~deb10u6+rpi1 [194 kB]
Get:5 http://quantum-mirror.hu/mirrors/pub/raspbian/raspbian buster/main armhf libsystemd0 armhf 241-7~deb10u6+rpi1 [305 kB]
Get:6 http://quantum-mirror.hu/mirrors/pub/raspbian/raspbian buster/main armhf systemd armhf 241-7~deb10u6+rpi1 [3,252 kB]
[... 1724 lines cut ...]
 /usr/bin/install -c -m 644 src/bluetooth.conf '/etc/dbus-1/system.d'
 /usr/bin/mkdir -p '/usr/share/dbus-1/services'
 /usr/bin/install -c -m 644 obexd/src/org.bluez.obex.service '/usr/share/dbus-1/services'
 /usr/bin/mkdir -p '/usr/share/dbus-1/system-services'
 /usr/bin/install -c -m 644 src/org.bluez.service '/usr/share/dbus-1/system-services'
 /usr/bin/mkdir -p '/usr/share/zsh/site-functions'
 /usr/bin/install -c -m 644 completion/zsh/_bluetoothctl '/usr/share/zsh/site-functions'
 /usr/bin/mkdir -p '/usr/share/man/man1'
 /usr/bin/install -c -m 644 tools/rctest.1 tools/l2ping.1 tools/bccmd.1 tools/btattach.1 '/usr/share/man/man1'
 /usr/bin/mkdir -p '/usr/share/man/man8'
 /usr/bin/install -c -m 644 src/bluetoothd.8 '/usr/share/man/man8'
 /usr/bin/mkdir -p '/lib/systemd/system'
 /usr/bin/install -c -m 644 src/bluetooth.service '/lib/systemd/system'
 /usr/bin/mkdir -p '/usr/lib/systemd/user'
 /usr/bin/install -c -m 644 obexd/src/obex.service '/usr/lib/systemd/user'
Synchronizing state of bluetooth.service with SysV service script with /lib/systemd/systemd-sysv-install.
Executing: /lib/systemd/systemd-sysv-install enable bluetooth
Created symlink /etc/systemd/system/dbus-org.bluez.service ΓåÆ /lib/systemd/system/bluetooth.service.
Created symlink /etc/systemd/system/bluetooth.target.wants/bluetooth.service ΓåÆ /lib/systemd/system/bluetooth.service.
Created symlink /etc/systemd/system/multi-user.target.wants/remapper.service ΓåÆ /lib/systemd/system/remapper.service.
sudo: unable to resolve host bthidhub: Name or service not known
Remote side unexpectedly closed network connection
FATAL ERROR: Remote side unexpectedly closed network connection

```

Dreamsorcerer, 2021-05-11:
I got the same error. I'm confused where the error comes from though, as it appears that every line of the setup script ran successfully, except `sudo reboot`. After triggering the reboot, it seems that everything runs.

Dreamsorcerer, 2025-01-08:
The error is actually non-fatal. The ssh disconnection is probably caused by the reboot itself.
Regardless, I've fixed the error from being emitted on the latest version by changing the order of commands.

closed 2025-01-08 by Dreamsorcerer