ruundii/bthidhub #21: Laptop paired as device

Dreamsorcerer, 2021-05-11:
I paired a laptop, but it shows as 'Paired' rather than 'Paired Host'. When I click the connect button, it connects to the laptop, then outputs (in CLI) 'Operation currently not available' and disconnects again.

Dreamsorcerer, 2021-05-11:
The first laptop I paired works correctly though, so not sure why this one has paired incorrectly, I followed the same steps for both, and both are running Ubuntu 20.04.

Dreamsorcerer, 2021-05-15:
My phone is producing the same result as well. It seems like only the first paired laptop works as a host...

The bluez devices for the 2 not working look like:
```
{'org.freedesktop.DBus.Introspectable': {}, 'org.bluez.Device1': {'Address': GLib.Variant('s', '88:DA:1A:B6:B5:D9'), 'AddressType': GLib.Variant('s', 'public'), 'Name': GLib.Variant('s', 's-phone'), 'Alias': GLib.Variant('s', 's-phone'), 'Class': GLib.Variant('u', 1835280), 'Icon': GLib.Variant('s', 'computer'), 'Paired': GLib.Variant('b', true), 'Trusted': GLib.Variant('b', false), 'Blocked': GLib.Variant('b', false), 'LegacyPairing': GLib.Variant('b', false), 'Connected': GLib.Variant('b', false), 'UUIDs': GLib.Variant('as', ['00001104-0000-1000-8000-00805f9b34fb', '00001105-0000-1000-8000-00805f9b34fb', '00001106-0000-1000-8000-00805f9b34fb', '00001108-0000-1000-8000-00805f9b34fb', '0000110a-0000-1000-8000-00805f9b34fb', '0000110b-0000-1000-8000-00805f9b34fb', '0000110c-0000-1000-8000-00805f9b34fb', '0000110e-0000-1000-8000-00805f9b34fb', '00001112-0000-1000-8000-00805f9b34fb', '0000112f-0000-1000-8000-00805f9b34fb', '00001132-0000-1000-8000-00805f9b34fb', '00001133-0000-1000-8000-00805f9b34fb', '00001200-0000-1000-8000-00805f9b34fb', '00005005-0000-1000-8000-0002ee000001']), 'Modalias': GLib.Variant('s', 'usb:v1D6Bp0246d0532'), 'Adapter': GLib.Variant('o', '/org/bluez/hci0'), 'ServicesResolved': GLib.Variant('b', false)}, 'org.freedesktop.DBus.Properties': {}}
```
and
```
{'org.freedesktop.DBus.Introspectable': {}, 'org.bluez.Device1': {'Address': GLib.Variant('s', '00:C2:C6:C0:44:2C'), 'AddressType': GLib.Variant('s', 'public'), 'Name': GLib.Variant('s', 's-laptop'), 'Alias': GLib.Variant('s', 's-laptop'), 'Class': GLib.Variant('u', 1835276), 'Icon': GLib.Variant('s', 'computer'), 'Paired': GLib.Variant('b', true), 'Trusted': GLib.Variant('b', false), 'Blocked': GLib.Variant('b', false), 'LegacyPairing': GLib.Variant('b', false), 'Connected': GLib.Variant('b', false), 'UUIDs': GLib.Variant('as', ['00001104-0000-1000-8000-00805f9b34fb', '00001105-0000-1000-8000-00805f9b34fb', '00001106-0000-1000-8000-00805f9b34fb', '00001108-0000-1000-8000-00805f9b34fb', '0000110a-0000-1000-8000-00805f9b34fb', '0000110b-0000-1000-8000-00805f9b34fb', '0000110c-0000-1000-8000-00805f9b34fb', '0000110e-0000-1000-8000-00805f9b34fb', '00001112-0000-1000-8000-00805f9b34fb', '0000112f-0000-1000-8000-00805f9b34fb', '00001132-0000-1000-8000-00805f9b34fb', '00001133-0000-1000-8000-00805f9b34fb', '00001200-0000-1000-8000-00805f9b34fb', '00001800-0000-1000-8000-00805f9b34fb', '00001801-0000-1000-8000-00805f9b34fb', '00005005-0000-1000-8000-0002ee000001']), 'Modalias': GLib.Variant('s', 'usb:v1D6Bp0246d0535'), 'Adapter': GLib.Variant('o', '/org/bluez/hci0'), 'ServicesResolved': GLib.Variant('b', false)}, 'org.freedesktop.DBus.Properties': {}}
```

Dreamsorcerer, 2021-06-26:
info from bluetoothctl for the laptop:
```
Device 00:C2:C6:C0:44:2C (public)
	Name: s-laptop
	Alias: s-laptop
	Class: 0x001c010c
	Icon: computer
	Paired: yes
	Trusted: no
	Blocked: no
	Connected: no
	LegacyPairing: no
	UUID: IrMC Sync                 (00001104-0000-1000-8000-00805f9b34fb)
	UUID: OBEX Object Push          (00001105-0000-1000-8000-00805f9b34fb)
	UUID: OBEX File Transfer        (00001106-0000-1000-8000-00805f9b34fb)
	UUID: Headset                   (00001108-0000-1000-8000-00805f9b34fb)
	UUID: Audio Source              (0000110a-0000-1000-8000-00805f9b34fb)
	UUID: Audio Sink                (0000110b-0000-1000-8000-00805f9b34fb)
	UUID: A/V Remote Control Target (0000110c-0000-1000-8000-00805f9b34fb)
	UUID: A/V Remote Control        (0000110e-0000-1000-8000-00805f9b34fb)
	UUID: Headset AG                (00001112-0000-1000-8000-00805f9b34fb)
	UUID: Phonebook Access Server   (0000112f-0000-1000-8000-00805f9b34fb)
	UUID: Message Access Server     (00001132-0000-1000-8000-00805f9b34fb)
	UUID: Message Notification Se.. (00001133-0000-1000-8000-00805f9b34fb)
	UUID: PnP Information           (00001200-0000-1000-8000-00805f9b34fb)
	UUID: Generic Access Profile    (00001800-0000-1000-8000-00805f9b34fb)
	UUID: Generic Attribute Profile (00001801-0000-1000-8000-00805f9b34fb)
	UUID: Vendor specific           (00005005-0000-1000-8000-0002ee000001)
	Modalias: usb:v1D6Bp0246d0535
```

It seems to be able to tell it is a computer, as the icon is set to 'computer'.

Dreamsorcerer, 2021-07-05:
Over the weekend it seems to have forgotten my first laptop. After pairing again, it's no longer a host, just like the other 2.
I now have exactly 0 devices that the RPi will send events to.

BaumDev, 2024-09-15:
Maybe a bit late, but any updates on this problem?