ruundii/bthidhub #28: Controller support?

Zamua, 2021-06-26:
Hello,

How much effort would it take for this to support standard controllers such as the DualShock 4 or dual sense? It would be nice to pair a pi zero w to my Apple TV/fire tv cube/google tv simultaneously and switch between them easily when playing games. Could make jumping between emulators on android and Apple TV games much more pleasant

great project!

Dreamsorcerer, 2021-06-26:
I don't think it really matters what kind of device it is, but you currently need to manually update sdp_record.xml to get most devices working correctly.
Try my rough instructions here and come back if you're struggling: https://github.com/ruundii/bthidhub/issues/11#issuecomment-841819628

Dreamsorcerer, 2025-01-08:
SDP in now automatically updated when a device is connected, so hopefully things will just work. Please reinstall the latest version and let us know if controllers work for you.