ruundii/bthidhub #30: MX Master 3 not an HID device

Dreamsorcerer, 2021-06-26:
After connecting a Logitech MX Master 3 to the device, it does not appear as an HID device for some reason...

I can see in `bluetoothctl` that it is an HID device:
```
[MX Master 3]# info
Device EA:2F:9D:84:5A:7D (random)
	Name: MX Master 3
	Alias: MX Master 3
	Appearance: 0x03c2
	Icon: input-mouse
	Paired: yes
	Trusted: no
	Blocked: no
	Connected: yes
	LegacyPairing: no
	UUID: Generic Access Profile    (00001800-0000-1000-8000-00805f9b34fb)
	UUID: Generic Attribute Profile (00001801-0000-1000-8000-00805f9b34fb)
	UUID: Device Information        (0000180a-0000-1000-8000-00805f9b34fb)
	UUID: Battery Service           (0000180f-0000-1000-8000-00805f9b34fb)
	UUID: Human Interface Device    (00001812-0000-1000-8000-00805f9b34fb)
	UUID: Vendor specific           (00010000-0000-1000-8000-011f2000046d)
	RSSI: -67
	TxPower: 4
```

But, it does not appear under `hid-recorder` or in the web interface.

jamezwhocares, 2021-11-08:
Same issues here. @Dreamsorcerer have you found a workaround?

Dreamsorcerer, 2021-11-08:
No, my entire system is currently unusable as it can no longer detect any of my machines as host devices.

Dreamsorcerer, 2025-01-08:
Everything should just work now on the latest version. No idea what that error was, but everything seems to be fine now and SDP is automatically updated as each device is added.

closed 2025-01-08 by Dreamsorcerer