ruundii/bthidhub #31: USB mouse and Magic Trackpad 1 not working

robgmills, 2021-08-02:
Super excited for this project!  Thank you for sharing it.

_**TL;DR -** Can you give me a bit more of a hint about what to look for and change in `sdp_record.xml`?_

I've successfully paired the Bluetooth hub with my MBP.  It's listed as a `Paired Host` in the web console and events from my keyboard are making it to my MBP via the Bluetooth hub. 

I'm having trouble getting my Magic Trackpad 1st Gen to work.  During the course of debugging, I've also been unsuccessful at getting a generic USB mouse (plugged into the Pi) to work.  Neither the track pad nor them mouse seems to be sending events to my host Macbook Pro (MBP) via the Bluetooth hub.  Both devices work when plugged directly into the MBP.

I've commented out [lines 10 and 11 in the `MouseMessageFilter`](https://github.com/ruundii/bthidhub/blob/master/mouse_message_filter.py#L10-L11) as [suggested](https://github.com/ruundii/bthidhub/issues/11#issuecomment-841819628) by @Dreamsorcerer in #11. 

I'm not sure how to edit the `sdp_record.xml` in order to get it to work.

I've installed `hid-tools` on the Pi and have successfully run `hid-recorder` on the paired track pad.  Here is the output:

```
# robgmills’s Trackpad
# 0x05, 0x01,                    // Usage Page (Generic Desktop)        0
# 0x09, 0x02,                    // Usage (Mouse)                       2
# 0xa1, 0x01,                    // Collection (Application)            4
# 0x85, 0x02,                    //  Report ID (2)                      6
# 0x05, 0x09,                    //  Usage Page (Button)                8
# 0x19, 0x01,                    //  Usage Minimum (1)                  10
# 0x29, 0x02,                    //  Usage Maximum (2)                  12
# 0x15, 0x00,                    //  Logical Minimum (0)                14
# 0x25, 0x01,                    //  Logical Maximum (1)                16
# 0x95, 0x02,                    //  Report Count (2)                   18
# 0x75, 0x01,                    //  Report Size (1)                    20
# 0x81, 0x02,                    //  Input (Data,Var,Abs)               22
# 0x95, 0x01,                    //  Report Count (1)                   24
# 0x75, 0x06,                    //  Report Size (6)                    26
# 0x81, 0x03,                    //  Input (Cnst,Var,Abs)               28
# 0x05, 0x01,                    //  Usage Page (Generic Desktop)       30
# 0x09, 0x01,                    //  Usage (Pointer)                    32
# 0xa1, 0x00,                    //  Collection (Physical)              34
# 0x15, 0x81,                    //   Logical Minimum (-127)            36
# 0x25, 0x7f,                    //   Logical Maximum (127)             38
# 0x09, 0x30,                    //   Usage (X)                         40
# 0x09, 0x31,                    //   Usage (Y)                         42
# 0x75, 0x08,                    //   Report Size (8)                   44
# 0x95, 0x02,                    //   Report Count (2)                  46
# 0x81, 0x06,                    //   Input (Data,Var,Rel)              48
# 0xc0,                          //  End Collection                     50
# 0x05, 0x06,                    //  Usage Page (Generic Device Controls) 51
# 0x09, 0x20,                    //  Usage (Battery Strength)           53
# 0x85, 0x47,                    //  Report ID (71)                     55
# 0x15, 0x00,                    //  Logical Minimum (0)                57
# 0x25, 0x64,                    //  Logical Maximum (100)              59
# 0x75, 0x08,                    //  Report Size (8)                    61
# 0x95, 0x01,                    //  Report Count (1)                   63
# 0xb1, 0xa2,                    //  Feature (Data,Var,Abs,NoPref,Vol)  65
# 0x06, 0x02, 0xff,              //  Usage Page (Vendor Usage Page 0xff02) 67
# 0x09, 0x55,                    //  Usage (Vendor Usage 0x55)          70
# 0x85, 0x55,                    //  Report ID (85)                     72
# 0x15, 0x00,                    //  Logical Minimum (0)                74
# 0x26, 0xff, 0x00,              //  Logical Maximum (255)              76
# 0x75, 0x08,                    //  Report Size (8)                    79
# 0x95, 0x40,                    //  Report Count (64)                  81
# 0xb1, 0xa2,                    //  Feature (Data,Var,Abs,NoPref,Vol)  83
# 0x85, 0x13,                    //  Report ID (19)                     85
# 0x15, 0x00,                    //  Logical Minimum (0)                87
# 0x25, 0x01,                    //  Logical Maximum (1)                89
# 0x75, 0x01,                    //  Report Size (1)                    91
# 0x95, 0x01,                    //  Report Count (1)                   93
# 0x06, 0x01, 0xff,              //  Usage Page (Vendor Usage Page 0xff01) 95
# 0x09, 0x0a,                    //  Usage (Vendor Usage 0x0a)          98
# 0x81, 0x02,                    //  Input (Data,Var,Abs)               100
# 0x06, 0x01, 0xff,              //  Usage Page (Vendor Usage Page 0xff01) 102
# 0x09, 0x0c,                    //  Usage (Vendor Usage 0x0c)          105
# 0x81, 0x22,                    //  Input (Data,Var,Abs,NoPref)        107
# 0x75, 0x01,                    //  Report Size (1)                    109
[... 117 lines cut ...]
	<attribute id="0x020a">
		<boolean value="true" />
	</attribute>
	<attribute id="0x020b">
		<uint16 value="0x0100" />
	</attribute>
	<attribute id="0x020c">
		<uint16 value="0x0c80" />
	</attribute>
	<attribute id="0x020d">
		<boolean value="true" />
	</attribute>
	<attribute id="0x020e">
		<boolean value="false" />
	</attribute>
	<attribute id="0x020f">
		<uint16 value="0x0640" />
	</attribute>
	<attribute id="0x0210">
		<uint16 value="0x0320" />
	</attribute>
</record>
```

I'm running macOS 11.5.1 "Big Sur" and Raspberry OS "buster."

Dreamsorcerer, 2021-08-02:
If it's listed as paired host, and you're connected to the laptop, then there should definitely be something in `hid-recorder` on the host. Admittedly, I've only tried on a Linux machine myself. This suggests that the host will not receive anything from the hub (or maybe you just forgot to connect the hub in BT settings?)...

The 2 things we are working with is the big long number in sdp_record.xml and the corresponding number after `R:` in the hid-tools output. Basically, you need to copy the latter number (removing the spaces and ignoring the first number, so starting from 0501) into the number in sdp_record.xml.
    I really need to see the hid-tools output to figure out where (and I have 0 paired hosts to test it myself right now...). But, the output would show what part is currently mapped to a mouse (it would be somewhat similar to your other output, so we'd probably be looking a section starting with 0902). That section of the number would be removed and replaced with the other one.

As your device already has multiple report IDs, you'd also need an additional change to the mouse filter code. I think your filter code would just end up as a single line:
`return b'\xa1' + msg`

Hopefully, if @ruundii comes back around to fix up some things, then we can look at making this whole process completely automatic (#23).

Dreamsorcerer, 2021-08-02:
If I were to take a guess from the info above, I'd say remove:
`05010902a1010901a100850305091901290815002501950875018102950081030600ff0940950275081581257f8102050109381581257f750895018106093009311601f826ff07750c95028106c0`
And at exactly the same place, add:
`05010902a10185020509190129021500250195027501810295017506810305010901a1001581257f09300931750895028106c00506092085471500256475089501b1a20602ff09558555150026ff0075089540b1a2851315002501750195010601ff090a81020601ff090c8122750195068101c0`

Dreamsorcerer, 2021-08-02:
If that doesn't work, it'll be pretty much impossible to debug without comparing hid-tools output on the host..

robgmills, 2021-08-03:
Thank you for your response!  You've given me some direction to dig in and debug further.  

It seems like macOS treats HID devices a bit differently than `hid-tools` expects.  I'll try to find the Report ID for the track pad on my macOS host a different way.  

I'll attempt to update the `sdp_record.xml` as you suggested and will report back.

robgmills, 2021-08-03:
I tried replacing the string recommended.  While it didn't exactly work, I'm now seeing Bluetooth packets that resemble the track pad events on the macOS host:

[image]

I'm sniffing these packets on the macOS host using the `PacketLogger.app` found in [_Additional Tools for XCode_](https://developer.apple.com/download/all/?q=additional%20tools).

Here's an example of me pressing `Command + Tab` on the keyboard paired with the hub and sniffed on the macOS host:

[image]

The image above shows four packets that represent, in order:

0. Pressing `Command`
1. Pressing `Tab`
2. Releasing `Tab`
3. Releasing `Command`

And here's the first sniffed packet from me scrolling on the track pad:

[image]

I'm still not sure why the keyboard works but the track pad doesn't but this is promising! Tomorrow I'll try to capture the packets when the connection is made between the hub and host.  I hope this will tell something about how macOS is registering the Bluetooth device.

Thank you again for pointing me in a direction!

robgmills, 2021-08-03:
Last update for the day:

I changed the _Report Filter_ for the track pad from `Mouse` to `Default`.  That resulted in the captured packets appearing like so:

[image]

The three red/brown/orange(?) packets are from when I attempted to three-finger scroll.  The rest of the packets appear to be recognized as a different type of packet that matches the working keyboard packets.  Unfortunately, the keyboard does not work anymore.

Dreamsorcerer, 2021-08-03:
Yes, stick to the modified Mouse filter. When you run `hid-recorder` on the Pi, it will show events from the mouse. Do the events look similar to the ones you are sniffing on the host? If so, compare some events (like a mouse click) and check exactly what the differences are.

Dreamsorcerer, 2025-01-08:
SDP records are now automatically updated, so everything should pretty much just work now. Please reinstall the latest version and give that a try.