ruundii/bthidhub #32: [Feature request] Wired host mode

asimonf, 2021-08-02:
I know Raspikey exists and this is supposed to be a wireless alternative. However, Raspikey only works with keyboards. This tool is much more useful in my scenario, but I need wired support rather than wireless.

Dreamsorcerer, 2021-08-02:
I suspect nobody is going to volunteer to work on such a feature. You'd probably find it easier to hack raspikey to use a different HID descriptor that work with a mouse.

asimonf, 2021-08-02:
I suspect that's true. I'm already giving it a try. Though I don't own a RPi Zero, I noticed that the code doesn't really have anything too specific to it and should work on debian with few modifications. I'm trying it on a Beaglebone right now with a USB BT adapter and since that's what I have in mind.

Dreamsorcerer, 2021-08-02:
Indeed, it's only really the automated install that likely requires a RPi Zero to work out of the box.

technikjunge, 2022-03-04:
I would also find this feature very useful.  How did you implement this exactly @asimonf ? Did you publish the code?

closed 2025-01-08 by Dreamsorcerer