ruundii/bthidhub #34: Question: Switching hosts on non-apple keyboards

Anonymous1893, 2021-08-11:
Does anyone know how to switch devices using this hub?

I can't find it anywhere in the documentation.

Dreamsorcerer, 2021-08-11:
Once I can actually use the project again, I'll look at making this customisable. If you're familiar with Python, it shouldn't be difficult to look through the code and put in some kind of hack in the meantime.

Anonymous1893, 2021-08-15:
Honestly don’t have much experience with large scale python stuff like this, but if i come up with a hack i’ll be sure to share it here. Appreciate you carrying this repo while the dev is MIA.

Dreamsorcerer, 2021-08-15:
Not sure I'd call it large scale (probably <1000 lines). :P

At a quick glance, the switch happens here:
https://github.com/ruundii/bthidhub/blob/c979471363b02e7ad7075dcb1a401cdbd8a81aaa/hid_devices.py#L69

If you print out the value of `tm` when pressing the desired button, you can probably just copy that value into the if statement and get it working... in theory...

Dreamsorcerer, 2025-01-08:
https://github.com/Dreamsorcerer/bthidhub/issues/12

closed 2025-01-08 by Dreamsorcerer