ruundii/bthidhub #35: No BT/ web GUI after installation on DietPi

robert-winkler, 2021-09-11:
Hi, I tried to install bthidhub on an RPi 0W with DietPi (<https://dietpi.com/>).
My final goal is to use the RPi 0W as a portable Stenography keyboard interface; Somehow inspired by the Stenogotchi <https://github.com/Anodynous/stenogotchi>. bthidhub seems more flexible for the connection of other devices than other options I found.
I cloned the GitHub code I slightly modified the `on_pi_setup.sh`: I substituted the `/home/pi/` by `/home/dietpi`, and run the script as root. Everything looks good and the Pi reboots.
Bluetooth is ON in the configuration of DietPi.

Now I wanted to try the Bluetooth capabilities and connect my keyboard.
However, after the reboot, 
1. I cannot see a Pi BT device on other computers.
2. I tried to open the web interface 127.0.1.1/bthidhub on the Pi and I get the following message:
`Error retrieving accessibility bus address: org.freedesktop.DBus.Error.ServiceUnknown; The name org.a11y.Bus was not provided by any .service file.`

OK, I got rid of this error by `apt install at-spi2-core`
and tried again connecting to the web interface. Now I get `BadURL`

Just to make sure the IP: `hostname -I` gives `192.168.1.196`; but connecting to this IP/bthidhub gives the same failure.

Any idea what is missing? 

How can I track down the problem and fix it?

Best regards, Robert

Dreamsorcerer, 2021-09-11:
I assume the `BadURL` error is from the Pi? i.e. with `systemctl status remapper` or similar.
How about pasting the full error?

Also, why are you trying to open the web interface via 127.0.1.1/bthidhub? The instructions say to do this via the Pi address, and there's no path (192.168.1.196:8080). Maybe that's all the issue is...

robert-winkler, 2021-09-14:
Ok, I tried again; step-by-step. I documented the complete procedure on a different computer and logged the output of the setup script and the remapper service (attached).
 
# Installation and testing
1. activate BlueTooth in the DietPi Config Module. This installs `pi-bluetooth bluez bluez-firmware` and installs `.service` files.
2. reboot.
3. `sudo su && git clone https://github.com/ruundii/bthidhub`.
4. `cd install/on_rpi/`.
5. `vim on_pi_setup.sh`.
6. adapt script to account for name of home directory: `:%s/home\/pi/home\/dietpi/g`.
7. I `dd` line `sudo reboot` to see final messages/errors.
8. `bash on_pi_setup.sh > /home/dietpi/bthidhub_setup.log`
9. I monitored the progress on a different terminal `watch tail bthidhub_setup.log`
Warnings/Errors during setup:
`Failed to connect to bus: No such file or directory` (several times).
`..pulseaudio.socket/service is masked, ignoring`.
10. Compilation/installations seems to work fine.
11. Reboot.
12. Trying to connect to web UI at `192.168.1.196:8080` and `localhost:8080`: Connection refused.
13. `systemctl status remapper`: Error.
14. The hostname was changed to bthidhub.
15. `sudo su` and `python3 ./remapper.py`
16. Connection to `192.168.1.196:8080` works!
17. but only locally; potential problem: on DietPi the standard user is called `dietpi`. 
18. I tested various browsers: the web UI does not work well with links2, w3m, uzbl, and nsgtk3 (lightweight browsers suitable for the Pi 0W); we need javascript, I suppose?
19. I installed the chromium browser and tested again the web UI. Now, the web UI looks nice, but it requests a login for the user 'pi'; This user does not exist on my DietPi. This is where I am stuck now.
  

# Remaining questions
- Any idea why the remapper service starts manually, but no automatically as service?
- Where could I change the username of the web UI?

# Log files:
[bthidhub_setup.log](https://github.com/ruundii/bthidhub/files/7162083/bthidhub_setup.log)
[remapper.log](https://github.com/ruundii/bthidhub/files/7162084/remapper.log)

Dreamsorcerer, 2021-09-14:
>     18. I tested various browsers: the web UI does not work well with links2, w3m, uzbl, and nsgtk3 (lightweight browsers suitable for the Pi 0W); we need javascript, I suppose?

Yes, I think it's all websockets. I can't even imagine how horrible the UX would be if done without JS (like asking the user to refresh the page every few seconds to see discovered BT devices).

>     * Any idea why the remapper service starts manually, but no automatically as service?

Again, there's no logs. Can't guess blindly.
If I run `systemctl status remapper` I can see error logs etc. at the bottom of the output. Your remapper.log file only shows the status fields from the top of the output, no error logs from the execution.
Maybe also running the exact same command as root might help reproduce the issue: https://github.com/ruundii/bthidhub/blob/master/install/on_rpi/remapper.service#L7
```
cd /home/pi/bthidhub
/usr/bin/python3 /home/pi/bthidhub/remapper.py
```

And now that I look at that, I'm wondering if you changed the paths in that file? Maybe I can guess blindly. :P

>     * Where could I change the username of the web UI?

At a guess, probably this one:
https://github.com/ruundii/bthidhub/blob/master/web.py#L19

robert-winkler, 2021-09-14:
You were completely right!

I changed two lines in the `remapper.py` from `/home/pi` to `/home/dietpi`. This modification enabled the remapper service.

In `web.py`, I changed
`PI_USER = 'pi'` to
`PI_USER = 'dietpi'` 
and the login works, also from remote machines.

Thanks a lot for your advice!

closed 2021-09-14 by robert-winkler