ruundii/bthidhub #38: Logitech M185/M225 doesn't work on MacOS

its-helvi, 2021-12-08:
Hi! I'm a high school student trying to use bthidhub to use an HP keyboard and a Logitech wireless mouse with my MacBook Air using a headless Raspberry Pi 3B+. The keyboard works almost perfectly; turning on caps lock illuminates the num lock light, which doesn't turn off until the raspi is restarted. The issue I have is with the mouse; I'm using a Logitech M225 with bthidhub, and my MacBook doesn't register any mouse movement, clicks or scrolls. I've tried using a different wireless mouse, USB port, and compatibility modes; nothing worked. 
Checking `sudo hid-recorder` shows that the raspi recognizes mouse clicks and scrolls (Wheel manipulates between 1 and -1), but the values for X and Y seem to reset to (0, 0) after every event. Turning the mouse off causes the cursor on my Mac to jump about 100 pixels towards the bottom right corner of the screen. 
I tried following the workaround in issue #11, but I have no idea where the mouse part or the Report ID is in sdp_record.xml. I'm also using the Raspbian Buster Lite image (non-legacy) from the Raspberry Pi foundation's imaging tool; I don't know if this caused any problems, but I did get some dependency issues when installing PyGObject that I resolved manually. ~~this issue might also be something obvious i'm not catching, in which case sorry in advance lol~~
Help would be appreciated.

Dreamsorcerer, 2021-12-11:
There's some additional information on how to edit the sdp_record.xml in https://github.com/ruundii/bthidhub/issues/31.

If you need more help, you'll probably need to copy more information here (output from hid-recorder).

Dreamsorcerer, 2025-01-08:
SDP records are automatically updated now, so things should generally just work now. Please try installing the latest version.

closed 2025-01-08 by Dreamsorcerer