ruundii/bthidhub #39: A1644 (Magic Keyboard 2) major lag

dorbareket, 2021-12-30:
As a RaspiKey user, I was super excited to find this bthidhub - but I have a huge lag with this one.
if I type a word like "hi", it would take a few seconds and then press at once "hhhhhhhhhiiiiiiiiiiiii".
any idea?

dorbareket, 2021-12-30:
Ok I played a little longer-
If I type complete sentences, all the keystrokes are considered with about 1.5seconds delay.
This is sometimes cool, as I can type "abcde...z" and then look at the screen ghost-typing 😆

dorbareket, 2022-01-08:
Hi, some more insights-
I just tried to use an external Bluetooth dongle instead of the integrated RPi zero Bluetooth, and the result is the same, major lag

Dreamsorcerer, 2025-01-08:
A lot has improved, so I suggest retrying with the latest version now. You also didn't state whether you used the Zero W, as described in the README. Some users have tried with a Pi 2 and reported severe lag with that, so I don't think this will work with anything older than a Zero.

closed 2025-01-08 by Dreamsorcerer