ruundii/bthidhub #41: Change Keyboard Layout

Aaroneisele55, 2022-01-06:
Hello,
Is it possible to change the layout of the keyboard?
I have a keyboard wich has the "physical" (letters on the keys) layout of "qwertz" (a german keyboard),
and it gets recognized by my echo show 8 and my laptop as a "qwerty" (english) keyboard, and i want to change it to qwertz, as i currently must google every time i want to input a special character (e.g. the german Ä, Ü or Ö).
How do i change this in bthidhub, as my echo show 8 has no option to change the layout?

Dreamsorcerer, 2022-01-06:
It's going to be a lot easier to change the layout in your OS. But, if you want to try, you'll have to look at the included message filters (e.g. https://github.com/ruundii/bthidhub/blob/master/hid_message_filter.py) and switch the codes around in there.

Aaroneisele55, 2022-01-07:
How should i change them? Where do i find the code for my german qwertz keyboard?

Dreamsorcerer, 2022-01-07:
Just add a print statement and figure it out by trial and error.

closed 2023-03-25 by Aaroneisele55