ruundii/bthidhub #43: Ipad

gdog9093, 2022-01-24:
struggling to get this to work with an iPad. I am connected to iPad, and have all the devices checked for communication, but no signs of a mouse or keyboard input on the ipad.

jacobsamlarose, 2022-03-01:
> struggling to get this to work with an iPad. I am connected to iPad, and have all the devices checked for communication, but no signs of a mouse or keyboard input on the ipad.

Interesting. Is this an isolated case? I'm planning to experiment with this, a Pi Zero 2 and a couple of iPads...

Dreamsorcerer, 2022-03-02:
They didn't say whether they tested on another device, so it may be that the input devices were not configured properly. Unfortunately, it's a massive pain to setup correctly, see my comments elsewhere on this repo (e.g. https://github.com/ruundii/bthidhub/issues/45#issuecomment-1055708356).

Dreamsorcerer, 2025-01-08:
This should mostly just work now, so please try installing the latest version.

If the IPad does not show as "Paired Host" after pairing, then open a new issue. The logic now checks if it's a computer or phone, so not sure what type of device an IPad reports as, but this can easily be changed.

closed 2025-01-08 by Dreamsorcerer