ruundii/bthidhub #49: Having trouble making it work

555learnwithme555, 2022-04-25:
Hi, thanks a lot for this cool project. I started with the hope making it to work which would solve the greatest pain of unplugging the USB dongles switching between computers. The main problem was mapping keyword and mouse. While trying to fix this I also came across another similar project but it showed some rays of light as detected my mouse and keyboard, beyond that it was not working.

It will be great if all would come together and make the two project work.

Please find the GitHub project page:
rpi-kvm: (https://github.com/BLeeEZ/rpi-kvm)https://github.com/BLeeEZ/rpi-kvm

Dreamsorcerer, 2022-04-25:
Nobody has time for that. If somebody decides to make some PRs to bring any changes over, then we can look at merging them.

I am curious how it works though, this project seems to require a patched version of bluez, while that project does not.

555learnwithme555, 2022-04-25:
Hi @Dreamsorcerer , thanks for replying. It will be really nice if someone checks this out, as I am still in learning curve I am bit limited to do major changes.  I had raised a issue in rpi-kvm and you can check on my findings.

Dreamsorcerer, 2025-01-08:
I've got SDP records updating automatically now, so this project should mostly just work now.

Still happy to work with another project if somebody wants to put the time in to figure out how to merge them, but I'm expecting that to be unlikely.

closed 2025-01-08 by Dreamsorcerer

Dreamsorcerer, 2025-01-11:
I think there's also a fundamental difference in implementation. That other project is working at the input event level (evdev), which is what the compatibility mode in this project did in the past. This project works at the HID level below that. I would assume that rpi-kvm thus only provides very basic mouse and keyboard support. Whereas, with the recent changes I've made to make SDP records automatically generated, this should work with pretty much any device and includes all the advanced functionality of those devices.

e.g. I have 2 mice with keyboard functions (copy, paste, alt+tab) and a keyboard with a trackpad, various media controls and a volume slider. All of these work out of the box, pretty much identically to when they are connected to the host directly.

Dreamsorcerer, 2025-01-11:
> I am curious how it works though, this project seems to require a patched version of bluez, while that project does not.

Actually, I don't see anything on that project which suggests it supports BT peripherals? In which case I don't think there's anything to learn from it.