ruundii/bthidhub #52: Looks like it's working but nothing is working.

miranda, 2022-10-02:
I have this installed on a Raspberry PI 2B with a bluetooth 4.2 usb dongle.  Everything seems to be working with discovering and pairing.  I tried to capture multiple usb keyboards but none of them type through to my Mac.  Bluetooth HID Hub RPi device is paired with the computer.  Also paired my Magic Mouse to the hub and it also is connected in the web UI but will not move my cursor.  Checked the remapper service on the pi and systemctl says it is running.  Not sure what else to check at this point, any help would be appreciated, I would REALLY like this to work it would be so great.

miranda, 2022-10-02:
I figured out the problem.  Instead of pairing the Mac to the pi from the Mac's bluetooth menu, you have to pair it from bthidhub's scan list.  It will only say "paired" instead of "paired host" if you do it from the Mac.  Magic Mouse still doesn't work but I'm guessing that's because the filter thing being set to default or mouse just doesn't support the Magic Mouse.

closed 2022-10-02 by miranda

Dreamsorcerer, 2022-10-02:
The filters are the original authors trick to work with their hardware, not actually general purpose functions. I've left several tips around in open issues for manually updating things for specific hardware (someday, maybe we'll get this automated). For example: https://github.com/ruundii/bthidhub/issues/31#issuecomment-891195536