ruundii/bthidhub #54: Followed the instructions, I cannot connect via web interface

AGene1234, 2023-02-28:
Brand new to Pis, this is my first Pi and my first attempt with a Pi. I want to make my wired keyboard and wired mouse, wireless via a bluetooth usb hub because they will be on a little rolling table and your project is ideal. So I bought a Pi Zero W starter kit, I had a 32 GB Sandisk and a couple of powerbanks to exchange them and I followed all the instructions to the letter:

1**) in the wpa_supplicant.conf set your wifi network and password**
_- right click, edit with notepad and typed the ssid and password, triple checked for typos. (it connected I can see it on the network)_

**2) After flashing Raspberry Pi OS (32-bit) Lite (https://www.raspberrypi.org/downloads/raspberry-pi-os/) to SD card, open it as a drive and copy wpa_supplicant.conf and ssh file to the root of the boot partition**
_- I downloaded Raspberry Pi Imager (https://www.raspberrypi.com/software/) and flashed the microsd with the Raspberry Pi OS (32-bit) Lite, the program showed on its list during the first step. It also had a gear icon for more settings but I didn't touch anything there_

**3) Plug SD card into the RPi and start it**
_- fine so far, placed the card to the reader and plugged a charger_

**4) get an IP address of the newly booted RPi**
_- once connected to the router i used an app on the phone (android), where it showed the device correctly by its name and its ip_

**Windows:**
**5) install putty https://www.putty.org/**
_- I downloaded and installed it without doing anything else with it_
      
**6) edit install_windows.bat and enter this ip address <-- added the correct address**
_- right click, edit with notepad, typed the address, triple checked for typos_

**7) run install_windows.bat, follow prompts until complete** 
_- double cliked on the bat file, a cmd opened showed a few things and prompted me to chose between yes no and other this. i typed y and pressed enter. it showed a few more lines pretty quickly and it closed before i read anything_
      
**That last step could take an hour to complete.
When the RPi reboots, the LED will switch off once the service is ready to use.**
_- 2 hours in, the LED doesn't turn off to indicate that the hub is ready for use (that's what I understand when you say that the service is ready when the LED is off)_

**Finally, go to the [rpi-ip-address]:8080 for the web configuration interface
raspberry**
_- i type the ip with the port to chrome's address bar, but it doesn't connect. i tried from the phone as well with the same results. i have kaspersky internet security but it never blocked activity via a web interface on anything_

I double clicked again on the install_windows batch file, clicked on the cmd, to stop it from closing and it is saying: "Please note that SSH may not work until a valid user has been set up"

any advice? i'll remind that i am totally new to this, so could I have missed a step that I suppose to know?

AGene1234, 2023-03-07:
close my issue. i have a logitech mouse with extra keys and they might want extra work for all i know. i'll use virtualhere instead.

Charlyo, 2023-06-04:
@AGene1234 you can fix this issue by installing hid-tools directly from the gitlab master branch directly.

Dreamsorcerer, 2023-09-11:
I've updated everything for bullseye now, so if you reinstall it you should atleast get something running successfully now.

closed 2023-09-11 by Dreamsorcerer