ruundii/bthidhub #55: Packages missing + py import error

JonnesCranach, 2023-03-07:
Hi,

great project, I need to have a deeper look into it when I have time.
pip is running into errors installing cairo, so I installed it via apt.
I'll try to 

python3-cairo python3-cairo-dev libcairo2-dev libgirepository1.0-dev
(after removing them again pip build was successfully) ... weird
  
starting remapper.py or also web.py is resulting in

**remapper.py**
```
Traceback (most recent call last):
  File "/home/pi/bthidhub/remapper.py", line 10, in <module>
    from adapter import BluetoothAdapter
  File "/home/pi/bthidhub/adapter.py", line 7, in <module>
    from hid_devices import *
  File "/home/pi/bthidhub/hid_devices.py", line 16, in <module>
    from compatibility_device import CompatibilityModeDevice
  File "/home/pi/bthidhub/compatibility_device.py", line 2, in <module>
    from hidtools.uhid import UHIDDevice
  File "/usr/local/lib/python3.9/dist-packages/hidtools/uhid.py", line 21, in <module>
    import hidtools.hid
  File "/usr/local/lib/python3.9/dist-packages/hidtools/hid.py", line 26, in <module>
    from hidtools.hut import HUT, U8, U16, U32
  File "/usr/local/lib/python3.9/dist-packages/hidtools/hut.py", line 27, in <module>
    from typing import (
ImportError: cannot import name 'TypeAlias' from 'typing' (/usr/lib/python3.9/typing.py)
```

**web.py**
```
Traceback (most recent call last):
  File "/home/pi/bthidhub/web.py", line 6, in <module>
    from hid_devices import *
  File "/home/pi/bthidhub/hid_devices.py", line 16, in <module>
    from compatibility_device import CompatibilityModeDevice
  File "/home/pi/bthidhub/compatibility_device.py", line 2, in <module>
    from hidtools.uhid import UHIDDevice
  File "/usr/local/lib/python3.9/dist-packages/hidtools/uhid.py", line 21, in <module>
    import hidtools.hid
  File "/usr/local/lib/python3.9/dist-packages/hidtools/hid.py", line 26, in <module>
    from hidtools.hut import HUT, U8, U16, U32
  File "/usr/local/lib/python3.9/dist-packages/hidtools/hut.py", line 27, in <module>
    from typing import (
ImportError: cannot import name 'TypeAlias' from 'typing' (/usr/lib/python3.9/typing.py)
```

Thanks for help or hints

JonnesCranach, 2023-03-07:
using 3.1+ helps?

Dreamsorcerer, 2023-03-07:
hidtools==0.3.1? It was released only a little earlier, so may or may not have the same issue. I guess 0.2 is the version we were using when testing, but getting hidtools to fix their testing/packaging would avoid us needing to pin to a specific release.

lswee, 2023-08-04:
Can confirm that changing the line:
[https://github.com/ruundii/bthidhub/blob/391044c2868255b2da6d593f6fa4826fa3bb330c/install/on_rpi/on_pi_setup.sh#L36](https://github.com/ruundii/bthidhub/blob/391044c2868255b2da6d593f6fa4826fa3bb330c/install/on_rpi/on_pi_setup.sh#L36)

To:
`sudo pip3 install hid-tools==0.2`

Fixes the error.