ruundii/bthidhub #56: USB connection between Rasp Pi and Host?

gtusr, 2023-05-13:
I need to connect via Bluetooth to the Rasp Pi and then connect the Rasp Pi to host via USB since the host cannot support Bluetooth.

One might assume that this is possible because the Rasp Pi Zero supports USB OTG.  

So if this is not already implemented, would it be an easy feature to add?

Dreamsorcerer, 2023-05-13:
This whole project is about proxying through Bluetooth, so I don't think this would fit into the project well.
If you want to try and take a look and figure out a way to include it, we can always take a look at a PR. But, I don't expect it to be a simple job.

closed 2023-05-13 by Dreamsorcerer

gtusr, 2023-05-13:
Would be nice to leave this issue open as a feature request in case someone can find some inspiration in one of these projects:

https://randomnerdtutorials.com/raspberry-pi-zero-usb-keyboard-hid/

https://github.com/HeuristicPerson/bluetooth_2_hid

https://gist.github.com/gbaman/50b6cca61dd1c3f88f41?permalink_comment_id=2788054