ruundii/bthidhub #68: Does not working with keyboard with trackpad

Raraph84, 2023-10-25:
Hi!

I'm trying to get this project working with a Bluetooth keyboard with a trackpad. My keyboard is the Diafield Keyboard. Everything is working except for the trackpad. I've tested all the filters, and none of them are working. However, I can see that with the Mouse filter, the mouse cursor moves a few pixels in one corner of the screen.

I've tried modifying the long hex string of the sdp_record with that of the hid-recorder and adjusting the filter as described in https://github.com/ruundii/bthidhub/issues/31, but with no success.

Any ideas?

My output from hid-recorder:
```
Available devices:
/dev/hidraw0:   DIAFIELD Keyboard
Select the device event number [0-0]: 0
# DIAFIELD Keyboard
# 0x05, 0x01,                    // Usage Page (Generic Desktop)        0
# 0x09, 0x06,                    // Usage (Keyboard)                    2
# 0xa1, 0x01,                    // Collection (Application)            4
# 0x85, 0x01,                    //  Report ID (1)                      6
# 0x75, 0x01,                    //  Report Size (1)                    8
# 0x95, 0x08,                    //  Report Count (8)                   10
# 0x05, 0x07,                    //  Usage Page (Keyboard)              12
# 0x19, 0xe0,                    //  Usage Minimum (224)                14
# 0x29, 0xe7,                    //  Usage Maximum (231)                16
# 0x15, 0x00,                    //  Logical Minimum (0)                18
# 0x25, 0x01,                    //  Logical Maximum (1)                20
# 0x81, 0x02,                    //  Input (Data,Var,Abs)               22
# 0x95, 0x01,                    //  Report Count (1)                   24
# 0x75, 0x08,                    //  Report Size (8)                    26
# 0x81, 0x03,                    //  Input (Cnst,Var,Abs)               28
# 0x95, 0x05,                    //  Report Count (5)                   30
# 0x75, 0x01,                    //  Report Size (1)                    32
# 0x05, 0x08,                    //  Usage Page (LEDs)                  34
# 0x19, 0x01,                    //  Usage Minimum (1)                  36
# 0x29, 0x05,                    //  Usage Maximum (5)                  38
# 0x91, 0x02,                    //  Output (Data,Var,Abs)              40
# 0x95, 0x01,                    //  Report Count (1)                   42
# 0x75, 0x03,                    //  Report Size (3)                    44
# 0x91, 0x03,                    //  Output (Cnst,Var,Abs)              46
# 0x95, 0x06,                    //  Report Count (6)                   48
# 0x75, 0x08,                    //  Report Size (8)                    50
# 0x15, 0x00,                    //  Logical Minimum (0)                52
# 0x26, 0xff, 0x00,              //  Logical Maximum (255)              54
# 0x05, 0x07,                    //  Usage Page (Keyboard)              57
# 0x19, 0x00,                    //  Usage Minimum (0)                  59
# 0x29, 0xff,                    //  Usage Maximum (255)                61
# 0x81, 0x00,                    //  Input (Data,Arr,Abs)               63
# 0xc0,                          // End Collection                      65
# 0x05, 0x0c,                    // Usage Page (Consumer Devices)       66
# 0x09, 0x01,                    // Usage (Consumer Control)            68
# 0xa1, 0x01,                    // Collection (Application)            70
# 0x85, 0x02,                    //  Report ID (2)                      72
# 0x15, 0x00,                    //  Logical Minimum (0)                74
# 0x25, 0x01,                    //  Logical Maximum (1)                76
# 0x75, 0x01,                    //  Report Size (1)                    78
# 0x95, 0x17,                    //  Report Count (23)                  80
# 0x0a, 0x94, 0x01,              //  Usage (AL Local Machine Browser)   82
# 0x0a, 0x92, 0x01,              //  Usage (AL Calculator)              85
# 0x0a, 0xae, 0x01,              //  Usage (AL Keyboard Layout)         88
# 0x0a, 0x23, 0x02,              //  Usage (AC Home)                    91
# 0x0a, 0x8a, 0x01,              //  Usage (AL Email Reader)            94
# 0x0a, 0x21, 0x02,              //  Usage (AC Search)                  97
# 0x0a, 0x83, 0x01,              //  Usage (AL Consumer Control Config) 100
# 0x0a, 0x2a, 0x02,              //  Usage (AC Bookmarks)               103
# 0x0a, 0x96, 0x01,              //  Usage (AL Internet Browser)        106
# 0x0a, 0x07, 0x03,              //  Usage (Vendor Usage 0x307)         109
# 0x0a, 0x81, 0x01,              //  Usage (AL Launch Button Config Tool) 112
# 0x0a, 0xaa, 0x01,              //  Usage (AL Desktop)                 115
# 0x0a, 0x24, 0x02,              //  Usage (AC Back)                    118
# 0x09, 0xb8,                    //  Usage (Eject)                      121
# 0x09, 0x6f,                    //  Usage (Display Brightness Increment) 123
[... 79 lines cut ...]
# 0x81, 0x06,                    //   Input (Data,Var,Rel)              284
# 0x05, 0x0c,                    //   Usage Page (Consumer Devices)     286
# 0x0a, 0x38, 0x02,              //   Usage (AC Pan)                    288
# 0x95, 0x01,                    //   Report Count (1)                  291
# 0x81, 0x06,                    //   Input (Data,Var,Rel)              293
# 0xc0,                          //  End Collection                     295
# 0xc0,                          // End Collection                      296
# 0x05, 0x0c,                    // Usage Page (Consumer Devices)       297
# 0x09, 0x01,                    // Usage (Consumer Control)            299
# 0xa1, 0x01,                    // Collection (Application)            301
# 0x85, 0xff,                    //  Report ID (255)                    303
# 0x05, 0x06,                    //  Usage Page (Generic Device Controls) 305
# 0x95, 0x01,                    //  Report Count (1)                   307
# 0x75, 0x02,                    //  Report Size (2)                    309
# 0x19, 0x24,                    //  Usage Minimum (36)                 311
# 0x29, 0x26,                    //  Usage Maximum (38)                 313
# 0x81, 0x02,                    //  Input (Data,Var,Abs)               315
# 0x75, 0x06,                    //  Report Size (6)                    317
# 0x81, 0x01,                    //  Input (Cnst,Arr,Abs)               319
# 0xc0,                          // End Collection                      321
# 
R: 322 05 01 09 06 a1 01 85 01 75 01 95 08 05 07 19 e0 29 e7 15 00 25 01 81 02 95 01 75 08 81 03 95 05 75 01 05 08 19 01 29 05 91 02 95 01 75 03 91 03 95 06 75 08 15 00 26 ff 00 05 07 19 00 29 ff 81 00 c0 05 0c 09 01 a1 01 85 02 15 00 25 01 75 01 95 17 0a 94 01 0a 92 01 0a ae 01 0a 23 02 0a 8a 01 0a 21 02 0a 83 01 0a 2a 02 0a 96 01 0a 07 03 0a 81 01 0a aa 01 0a 24 02 09 b8 09 6f 09 70 09 b6 09 b5 09 30 09 cd 09 e9 09 ea 09 e2 81 02 95 01 75 01 81 03 c0 05 0c 09 01 a1 01 85 03 05 01 09 06 a1 02 05 06 09 20 15 00 26 ff 00 75 08 95 01 81 02 06 bc ff 0a ad bd 75 08 95 06 81 02 c0 c0 05 01 09 80 a1 01 85 04 15 00 25 01 75 01 95 03 19 81 29 83 81 02 95 01 75 05 81 03 c0 05 01 09 02 a1 01 85 05 09 01 a1 00 05 09 19 01 29 03 15 00 25 01 95 03 75 01 81 02 95 01 75 05 81 01 05 01 09 30 09 31 16 01 f8 26 ff 07 75 0c 95 02 81 06 09 38 15 81 25 7f 75 08 95 01 81 06 05 0c 0a 38 02 95 01 81 06 c0 c0 05 0c 09 01 a1 01 85 ff 05 06 95 01 75 02 19 24 29 26 81 02 75 06 81 01 c0
N: DIAFIELD Keyboard
I: 5 04e8 7021
```

Dreamsorcerer, 2023-10-25:
I'm afraid I don't have time currently to help. But, do you have multiple pointers connected?
I've found that only 1 pointer will work, and don't know how to fix it without needing to convert them all input pointer events into a single output pointer event (which is obviously more complex than just proxying the events untouched):
https://unix.stackexchange.com/questions/757291/hid-events-are-received-but-mouse-input-still-not-working

Dreamsorcerer, 2023-10-25:
No, wait, if you got some movement with the mouse filter, then that's not the case. You must have just got the SDP record wrong. When I get back to this, I think I may have enough information now to make this automated. So, will try and resolve this properly in a few weeks probably.

Raraph84, 2023-10-25:
Indeed, the filter is not suitable for my trackpad. I noticed that only 3 bytes change when I move the mouse. I've figured out how to create these bytes from scratch, and I can make the mouse cursor move by modifying the variables:
```
button = 0
x = 0
y = 0
scroll = 0
return bytes([
    0xA1,
    0x03,
    button,
    0x00,
    x if x >= 0 else (256 + x),
    0 if x >= 0 else 255,
    y if y >= 0 else (256 + y),
    0 if y >= 0 else 255,
    scroll
])
```
I'm not sure if this is the right way to do it, but now I'm trying to decode the bytes from my trackpad so that I can create a filter that works for my keyboard/trackpad.

Dreamsorcerer, 2023-10-25:
> the filter is not suitable for my trackpad

I don't think the filter is suitable for anything other than ruundii's mouse...

Dreamsorcerer, 2023-10-25:
As explained in the threads elsewhere, my approach is to remove filtering, and just update the SDP record to match the mouse's record. Then we can just pass events through without modification (or only needing to add/edit a byte for the report ID).

Raraph84, 2023-10-25:
Yes, as mentioned earlier, I tried by modifying the SDP, but it didn't solve the problem. I have no idea why, perhaps because the same device functions as both a keyboard and a trackpad at the same time.

Dreamsorcerer, 2023-10-26:
It would just have multiple report IDs, which would all need to be included and remapped.

Dreamsorcerer, 2025-01-08:
SDP records are now automatically updated, so everything should pretty much just work now. Please reinstall the latest version and give that a try.

closed 2025-01-08 by Dreamsorcerer