ruundii/bthidhub #69: Installation Instructions - Helpful Clarification (Adjust SH && BAT for Windows if any username change is made to defaults on Pi)

CanadianVice, 2023-11-10:
So I just noticed something that cost me a decent chunk of time (my bad) - if one configures the PI for headless during the flashing process, they may well be inclined to change username and such from the defaults.

My suggestion is one of either:
- Add to the Readme that you should keep the default settings when flashing the PI or the script defaults won't work
- Add to the readme a warning that if you changed the default account on the PI, not only is there the Windows.bat install modification to do, but one needs to change some paths in the setup.sh script as well to reflect /home/<user> rather than the default /home/pi

It would probably spare people some trouble. Not your fault, but clarification never hurts!

CanadianVice, 2023-11-10:
Right now the instructions only reference changing the Windows BAT which may lead to tunnel vision. Certainly why I didn't clue in xD

Dreamsorcerer, 2023-11-10:
Previous versions of Raspbian didn't have a config for username. I added some points about the username, but clearly didn't catch everything. Feel free to make a PR, we probably want to use a variable at the top of the scripts for the username, in order to have the username work correctly throughout the file.

CanadianVice, 2023-11-10:
Breaks the on_pi.sh script too, for the same reasons.

BaumDev, 2024-09-13:
Same problem in the remapper.service

Dreamsorcerer, 2025-01-08:
I'm not clear what (if any) issues still remain. Please open a PR with suggestions if they still do.

closed 2025-01-08 by Dreamsorcerer