ruundii/bthidhub #70: cant install, last code mypyc unknown command

rayayamam, 2023-11-17:
it gets through the installation process to the last part line before the reboot command.

# Compile some Python modules to reduce lag.
# We do this at the end, as the project is already usable without this step.
cd /home/pi/bthidhub/
mypyc

"mypyc unnown command"

never got it running, waited about 1 hour in. bullseye and bookworm lite image. same error and cant connect via web interface

Dreamsorcerer, 2023-11-17:
It's in the requirements file, so something must have failed to install it earlier?
https://github.com/ruundii/bthidhub/blob/master/requirements.txt#L9

rayayamam, 2023-11-17:
i currently trying to install it. i manually installed mypy and pip. now starting 

git clone https://github.com/ruundii/bthidhub
cd /home/pi/bthidhub/install/on_rpi
bash ./on_pi_setup.sh

hope it works.

rayayamam, 2023-11-17:
finally it worked. mypy may be got renamed or isnt up to date on the repo. manually installing it worked.

closed 2023-11-17 by rayayamam