ruundii/bthidhub #82: What is the web page or port to access configuration?

berkaysit, 2024-02-05:
This projects look amazing. But I cannot start using yet.

I installed on RPi 3. After the reboot, the taskbar on the desktop GUI is gone. I only see Wastebasket icon and wallpaper. As I understand there is a web page to configure this program, however, the address and port were not mentioned in any doc. I tried to reach the web page from laptop's browser by <RPi_IP>:80 but didn't work.

Can you please give a bit more information on how to access installed program?

Dreamsorcerer, 2024-02-05:
Sounds like you didn't follow the install instructions?
https://github.com/ruundii/bthidhub/tree/master/install

Dreamsorcerer, 2024-02-05:
> Sorry, I missed the last line :( I was connected to RPi by Putty, cloned the project, and executed :~/bthidhub/install/on_rpi $ sudo ./on_pi_setup.sh

Ah, OK, I was confused by your description of a desktop, as it's all designed to be run headless.

> Unfortunately, [rpi-ip-address]:8080 still does not work for me.

If the script didn't error out, and the LED switched off after booting, then it should be ready.
If not, you'll have to SSH in and take a look at what's going on. I forget the commands off the top of my head, but something like `systemctl status remapper` or `journalctl -xe` should give you information on whether the service is running or what errors it encountered.

closed 2025-01-08 by Dreamsorcerer