ruundii/bthidhub #94: Installation / mypyc ObjPath compatibility issues

wihrl, 2024-12-07:
Hi, I ran into trouble during installation with pip: 
```
error: externally-managed-environment

× This environment is externally managed
╰─> To install Python packages system-wide, try apt install
    python3-xyz, where xyz is the package you are trying to
    install.

    If you wish to install a non-Debian-packaged Python package,
    create a virtual environment using python3 -m venv path/to/venv.
    Then use path/to/venv/bin/python and path/to/venv/bin/pip. Make
    sure you have python3-full installed.

    For more information visit http://rptl.io/venv
```

I fixed this by installing the requirements manually using: `sudo pip3 install -r $HOME/bthidhub/requirements.txt --break-system-packages`

After this the project worked fine without mypyc, but was extremely slow - so i ran mypyc on it and now remapper.py no longer starts up:
```
× remapper.service - Remapper
     Loaded: loaded (/lib/systemd/system/remapper.service; enabled; preset: enabled)
     Active: failed (Result: exit-code) since Sat 2024-12-07 12:53:23 CET; 1min 2s ago
   Duration: 1.638s
    Process: 464 ExecStart=/usr/bin/python3 /home/pi/bthidhub/remapper.py (code=exited, status=1/FAILURE)
    Process: 465 ExecStartPost=/usr/bin/bash -c echo none > /sys/class/leds/ACT/trigger (code=exited, status=0/SUCCESS)
    Process: 480 ExecStopPost=/usr/bin/bash -c echo heartbeat > /sys/class/leds/ACT/trigger (code=exited, status=0/SUCC>
   Main PID: 464 (code=exited, status=1/FAILURE)
        CPU: 888ms

Dec 07 12:53:22 bthidhub python3[464]: Traceback (most recent call last):
Dec 07 12:53:22 bthidhub python3[464]:   File "/home/pi/bthidhub/remapper.py", line 10, in <module>
Dec 07 12:53:22 bthidhub python3[464]:     from adapter import BluetoothAdapter
Dec 07 12:53:22 bthidhub python3[464]:   File "adapter.py", line 11, in <module>
Dec 07 12:53:22 bthidhub python3[464]:     from agent import Action, Agent
Dec 07 12:53:22 bthidhub python3[464]:   File "agent.py", line 14, in <module>
Dec 07 12:53:22 bthidhub python3[464]:     device: dt.ObjPath
Dec 07 12:53:22 bthidhub python3[464]: KeyError: 'ObjPath'
Dec 07 12:53:23 bthidhub systemd[1]: remapper.service: Main process exited, code=exited, status=1/FAILURE
Dec 07 12:53:23 bthidhub systemd[1]: remapper.service: Failed with result 'exit-code'.
```

Any ideas on what could be causing this? Maybe a wrong package version?

wihrl, 2024-12-07:
Ok, i got it to work by downgrading mypy: `sudo pip3 install mypy==1.10.1 --break-system-packages`
But not sure if it actually compiled anything since it still pegs 1 core to 100% all the time and works really slowly.

Dreamsorcerer, 2024-12-07:
I think the last update was for bullseye, so I think we need to retest and upgrade the installation for bookworm.

Dreamsorcerer, 2025-01-08:
Done. Bookworm installation will work now.

closed 2025-01-08 by Dreamsorcerer